digidotcom/xbee-python #280: Need help with Xmodem

rahul007-bit, 2022-08-20:
Hi Everyone,
Currently, I am working on my university project were, we have to transfer an image via Zigbee XBee Module S2C.

So, I am trying to use the Xmodem (https://xbplib.readthedocs.io/en/latest/api/digi.xbee.util.xmodem.html) protocol but cannot understand how to use it.

#### Sender Code
``` 
def write_cb(Bytearray):
    try:
        device.send_data(remote_device, Bytearray)
        time.sleep(0.25)
        return True
    except Exception as e:
        print(e)
        return False

def read_cb(size_read, timeout):
    message = device.read_data_from(remote_device, timeout=timeout)
    print(message)
    return message
try:
    device.open()

    # Obtain the remote XBee device from the XBee network.
    xbee_network = device.get_network()
    remote_device = xbee_network.discover_device(REMOTE_NODE_ID)
    if remote_device is None:
        print("Could not find the remote device")
        exit(1)

    send_file_ymodem(FILE, write_cb=write_cb, read_cb=read_cb)

    print("Success")

finally:
    if device is not None and device.is_open():
        device.close()

```
#### Receiver Code
```    
def write_cb(Bytearray):
  try:
      res = device.send_data_async(remote_device,Bytearray)
      print(res)
      if res:
          return True
      else:
          return False
  except Exception as e:
      print(e)
      return False

def read_cb(size_read, timeout):
  try:
      size_read = device.read_data_from(remote_device, timeout=timeout)
      return size_read
  except Exception as e:
      print(e)
      return None

try:
    device.open()
    xbee_network = device.get_network()
    remote_device = xbee_network.discover_device(REMOTE_NODE_ID)
    if remote_device is None:
        print("Could not find the remote device")
        exit(1)

    get_file_ymodem(FILE, write_cb, read_cb)
    print("Waiting for data...\n")
    input()

finally:
    if device is not None and device.is_open():
        device.close()
```

This is what I have tried till now, you may have guessed till now it's not going to work, so please guide me 🙏🏼 🥺.

Your reply will be like thor's hammer for which I am worthy(I assume).

any example will help a lot.

Thank You,
waiting for your reply 🤗!!

sandeep1404, 2024-01-15:
Can you able to achive this?

rahul007-bit, 2024-01-19:
Yes, I was able to complete the project. I managed to transfer images or audio files using the Zigbee XBee Module S2C. However, I cannot recall the exact solution I used. If you would like to know my approach, please let me know.

sandeep1404, 2024-01-19:
Hi @rahul007-bit, That's great that you could transfer images using ZigBee, If u don't mind can u send the documentation or related work that you have done to transfer images using the xbee module, It would really help my project. My mail ID is : sandeeptechiot@gmail.com can you please guide me in this regard? It would be a great help for me.  Thank you.

rahul007-bit, 2024-01-19:
I can send you the file I used, but it would be difficult to provide the supporting documents since it has been a long time since I worked on this project.

rahul007-bit, 2024-01-19:
[Xbee.zip](https://github.com/digidotcom/xbee-python/files/13991002/Xbee.zip)

@sandeep1404

sandeep1404, 2024-01-19:
Thank you so much @rahul007-bit. for the help. Will check that and let you know if i get any issues. Thank you.

rahul007-bit, 2024-01-19:
Yeah, it was very very slow even for transferring a 100kb image, since this project just to prove to my group that we can transfer images. for testing, I used an image of 1kb image which was a very low resolution. if you want to transfer images over radio waves you also check [sstv](https://en.wikipedia.org/wiki/Slow-scan_television).