digidotcom/xbee-python #282: Problems getting the LQI value

fealvares, 2022-12-08:
After I run a deep discovery, I try to get the RSSI value from a node and its neighbours. Sometimes I get a value equals 9999, why this happen?

This is my code

```
remote_xbee = self.xnet.get_device_by_64(XBee64BitAddress.from_hex_string(frame['addr']))
        for i in self.xnet.get_node_connections(remote_xbee):
            addr_a = i.node_a.get_64bit_addr().address
            addr_b = i.node_b.get_64bit_addr().address
            lq = abs(i.lq_a2b.lq)
            rssi = int.to_bytes(lq, 1, 'big')
            payload = {'value': addr_a + rssi + addr_b, 'topic': FIND_PUBLISH_TOPIC}
            UP.put(payload)
```

In the __init__ I defined self.xnet = self.xbee.get_network()

frame['addr'] has a string with the node's address that I want to analyze.

So I get the link quality, coverts it to bytes and build my payload.

tatianaleon, 2022-12-12:
Hi @fealvares,

That value is for an unknown quality of the connection. That is, there might be a connection between node A and node B but with an unknown quality (LQI or RSSI, depending on the protocol you are using) in one of the connection directions. For example, the quality A > B is known but from B > A is unknown, because communication from B to A have never occurred but from A to B did.

Best Regards,
Tatiana

fealvares, 2022-12-12:
Hi @tatianaleon ,

Thanks for the answer.

Do you know if I force a communication between A and B I can get the correct RSSI value?

Best Regards,
Felipe

tatianaleon, 2022-12-12:
Hi @fealvares,

I guess, but that "forcing" is not so easy. If you are using a network protocol as Zigbee or DigiMesh, the mesh is going to select the way for each packet.

Best Regards,
Tatiana

fealvares, 2022-12-12:
Hi @tatianaleon,

I am using the DigiMesh protocol. I was thinking send a broadcast message when the RSSI is requested in a device, but as you wrote, the mesh will select the way automatically.

But if the radios are not communicate in my mmesh network doesn't make sense know thr RSSI value.

Thanks for clarifying.

Best Regards
Felipe

tatianaleon, 2022-12-13:
Closing as re

closed 2022-12-13 by tatianaleon