digidotcom/xbee-python #283: Node timeout doesn't work in my script

fealvares, 2022-12-08:
My discovery does not finish after exced the node timeout after discovering the last node.

My discovery configurations:
```
self.xnet = self.xbee.get_network()
        # discovery nodes settings
        self.xnet.set_deep_discovery_options(deep_mode=NeighborDiscoveryMode.CASCADE,
                                             del_not_discovered_nodes_in_last_scan=False)
        self.xnet.set_deep_discovery_timeouts(node_timeout=25, time_bw_requests=10, time_bw_scans=20)
```
Am I doing a misapllication?

The discovery is taking too much time to finish.

fealvares, 2022-12-12:
Hi @tatianaleon,

I am using the DigiMesh protocol. I was thinking send a broadcast message when the RSSI is requested in a device, but as you wrote, the mesh will select the way automatically.

But if the radios are not communicate in my mmesh network doesn't make sense know thr RSSI value.

Thanks for clarifying.

PS: I think you mistook your answer with the issue https://github.com/digidotcom/xbee-python/issues/282.

Best Regards,
Felipe

tatianaleon, 2022-12-12:
Hi @fealvares,

Yeah, I have both questions opened and... you know :-)
Adding the response to the right question. I will try to answer this one ASAP.

Best Regards,
Tatiana

tatianaleon, 2022-12-13:
Hi @fealvares,

I am not sure of what you mean with "_My discovery does not finish after exceed the node timeout after discovering the last node_" and later you add "_The discovery is taking too much time to finish_". What is exactly happening? Are you using the `deep` discovery? Are you registering a callback for the end of the process (`add_discovery_process_finished_callback()`)? Are you waiting for the process checking `is_discovery_running()`? The code example it is not complete.

The "deep" discovery process usually takes more time than the standard discovery process, although it offers more information about the network (connections, quality) and does not generate so much traffic. The time to discover the full network with this mechanism depends on the node timeout but it is not the only parameter involved. See [Deep discovery](https://xbplib.readthedocs.io/en/stable/user_doc/discovering_the_xbee_network.html#deep-discovery) for more information.
If you are using `start_discovery_process(deep=True)` in a DigiMesh network, with the configuration:
* mode: CASCADE
* node_timeout: 25s
* time_bw_requests: 10s
* time_bw_scans: 20s

The discovery will wait 25s per node to get its neighbors, 10s between node requests, and 20 seconds between scans (by default 1 scan if you are not passing `n_deep_scans=<n>` to `start_discovery_process`)
That is, if you have `N` nodes in your network, the time per discovery scan, `DS`, is:

`DS ~= N*25 + (N-1)*10`

And the total discovery time with `S` number of scans, `DT`, is:

`DT ~= S*DS + (S-1)*20`

For example, 2 scans with a network with 5 nodes:

```
DS ~= 5*25 + (5-1)*10 ~= 165s
DT ~= 2*165 + (2-1)*20 ~= 350s
```

Hope this helps.
Tatiana

fealvares, 2022-12-14:
Hi Tatiana,

I thought the discovery nodes would end when the scan reach the node timeout if it didn't find any nodes, so after discover a node and pass 25 seconds without discovering a new node I was expecting the end of the scan.

I'm registering a callback for the end of process, its prints in my log when the scan stops. My dev team requested it to show in our dashboard when the discover is over.

I understood how the deep discovery timeout works now, thanks. 

I'm doing some test with the discovery nodes in a farm that has more than 50 xbees, acutally are two customers farms that their mesh network are connected. Its take about 15 minutes to conclude the discovery :)

I run the deep scan to discorver my network, I do this beacuse it was the only way I found to get the link quality. Does the lib have other way to get the link quality? If I could get the link quality without the deep scan, I would use the simple discovery instead.

Att,
Felipe

tatianaleon, 2022-12-15:
Hi @fealvares 

> I thought the discovery nodes would end when the scan reach the node timeout if it didn't find any nodes, so after discover a node and pass 25 seconds without discovering a new node I was expecting the end of the scan.

You are right, if there is only one node in the network (your local node) and only one scan, the discovery will take that 25 seconds.

> I'm registering a callback for the end of process, its prints in my log when the scan stops. My dev team requested it to show in our dashboard when the discover is over.

Just for you to know, you can register callbacks for the [end of the discovery process](https://xbplib.readthedocs.io/en/stable/user_doc/discovering_the_xbee_network.html#notify-discovery-finishes), for [new discovered devices](https://xbplib.readthedocs.io/en/stable/user_doc/discovering_the_xbee_network.html#notify-new-discovered-nodes), but also for beginning and end of each scan or for [network modifications](https://xbplib.readthedocs.io/en/latest/user_doc/discovering_the_xbee_network.html#listen-to-network-modification-events) (see [Network modifications sample](https://github.com/digidotcom/xbee-python/tree/1.4.1/examples/network/NetworkModificationsSample))

> I'm doing some test with the discovery nodes in a farm that has more than 50 xbees, acutally are two customers farms that their mesh network are connected. Its take about 15 minutes to conclude the discovery :)

You can reduce your time values, but keep in mind this it is going to generate more traffic in the network.

> I run the deep scan to discorver my network, I do this beacuse it was the only way I found to get the link quality. Does the lib have other way to get the link quality? If I could get the link quality without the deep scan, I would use the simple discovery instead.

The library does not include any other way. You can use other methods such as the [`DB`](https://www.digi.com/resources/documentation/Digidocs/90002277/#Reference/r_cmd_DB.htm) or a  [test link](https://www.digi.com/resources/documentation/Digidocs/90002277/#Tasks/t_testing_adjacent_links.htm) but you have to implement it with the library API for your application and I am not sure if this will give you the quality of all the connections, since only the local node can perform them.

 I do not know why you need the whole network connections quality. It is useful to better position nodes on a deployment, to check the network health or to identify bottle necks in the communication, for example. I mean, if one node fails and the network is well setup, the network will find a different path to the destination (see [DigiMesh networking](https://www.digi.com/resources/documentation/Digidocs/90002277/#Concepts/c_mesh_networking.htm)) and in the mean time the "bad" node can be fixed.

[image]

Best Regards,
Tatiana

fealvares, 2022-12-15:
Hi Tatiana,

We need the network quality connections to our after sales team can indentify if a communication problem is related to radio or something else.

It's make sense not to be a quick response when I want to analyze my network.

I will change the deep discovery to the standard discovery, and also a new feature to do a deep scan only when it is necessary, I think it will be the better solution.

Thanks a lot for you help Tatiana.

Att,
Felipe

closed 2023-01-11 by tatianaleon