digidotcom/xbee-python #298: Error while trying to get 64-bit address of XBee

sayaz, 2024-01-19:
I am using Digi Xbee3 Zigbee mesh kit module. As of now I am using one node as Coordinator and other as End device. Both nodes are using the firmware [Digi XBee3 802.15.4 - 200D]. The specific reason for using this firmware is because I want to stay with MAC and PHY layer data. Also I want to have control on the transmission channel (CH).

I am using Python to send and receive packets. The communication works just fine. The only problem I am facing is, once the transmitter transmits, it is taking about 4 seconds for the receiver to print the receive message. To troubleshoot that I have printed the logs, one of which: (This has a timeout of 4 seconds.)

`DEBUG : 9789 - Error while trying to get 64-bit address of XBee (FFFFFFFFFFFFFFFF -  - 0000): Response not received in the configured timeout.`

After this, the transmission does happen and received data gets printed.

**Configuration**:

- MAC Mode: 802.15.4 + Digi header w/ACKs
- Sleep mode: No Sleep
- API Enable: API Mode Without Escapes

**Till now I did the below**:

1. Made sure the receiver has actually received the 64-bit address. I am able to get the 64-bit address `device.get_64bit_addr()` before the error occurs.
2. Since I am using the firmware 802.15.4, the classes I am using are :  `Raw802Device` and `RemoteRaw802Device`
3. I have used different firmware of [Digi XBee3 Zigbee 3.0 - 1013]. This version works perfectly fine (i.e. no delay).
4. I've tried to decrease the timeout option `device.set_sync_ops_timeout(1)`. This does not have any effect.

Does anyone have any hint on how to get a fix on this delay ?

rubenmoral, 2024-01-19:
Hi @sayaz,

What is the value of the MY parameter of your two XBee 3 devices? Note that the parameter's default value is 0x0, you must set 0xFFFF if you want to use the 64-bit addressing. Another option would be to set different MY values for each module to use 16-bit addressing.

See:

- https://www.digi.com/resources/documentation/digidocs/90002273/Default.htm#concepts/c_peer_to_peer.htm?TocPath=Networking%257C_____7
- https://www.digi.com/resources/documentation/digidocs/90002273/Default.htm#reference/r_cmd_my.htm

sayaz, 2024-01-19:
@rubenmoral Thank you so much for this. This solves the issue. I missued this link. Cheers.

closed 2024-01-19 by sayaz